Nova-UTD/hardware #3: Mount Ouster lidar interface box & wiring

jruths, 2023-10-26:
Nominally it seems like the box is powered with 24V DC, but I believe it should work with 12V DC... should double check!

meyli1, 2023-10-26:
OS! - Rev 7:  https://static.ouster.dev/sensor-docs/hw_user_manual_OS1/hw_common_sections_OS1/os1-overview.html#intro-os1

5408Andy, 2023-10-26:
Important Note regarding power supply for Interface Box: "The OS1 Sensors are designed to operate between 12V and 24V nominal input voltage. A low voltage warning will be triggered if the voltage at the sensor connector drops below 9.5V. The sensor will shut down if this input voltage drops to 9V. The maximum input voltage is 34V for the OS1 sensor."

jruths, 2023-10-31:
@meyli1 @5408Andy I know Sasha is out of town, so I just drilled these holes in the lidar box mounting plate. You should be good to go now!

closed 2024-02-01 by danielv012