Nova-UTD/hardware #5: OBC mount

jruths, 2023-10-26:


SashaLuich, 2023-11-17:
https://cometmail.sharepoint.com/sites/Voltron/_layouts/15/Doc.aspx?sourcedoc={f8358ecf-49d8-430c-83a8-a84b3b49a71b}&action=edit&wd=target%28HFE.one%7Cfb15938b-1acf-45b8-9dfa-7cf0c26b9a0c%2FOBC%20Mounts%7C31f95022-b050-fe48-b85a-3b3ed4ad9fc7%2F%29&wdorigin=NavigationUrl

5408Andy, 2023-12-07:
I have added some ideas on the the OneNote regarding implementation of OBC mount.

5408Andy, 2024-01-19:
Reply from OnLogic regarding orientation. The computer is rugged so orientation doesn't matter as long as we have it solidly mounted with some kind of metal plate. There should be no issue regarding implementing the OBC against the firewall

[image]
[image]