Nova-UTD/hardware #6: Design front rack (includes radar, LED matrix mounts)

jruths, 2023-10-26:


5408Andy, 2023-10-27:
Quick/Jank Solution: Keep the original mount with the push bar coming off the hood. To prevent the bar from swaying forward, we can use zip-tips mounted off the intake/heat-sync holes.

Abhi220180, 2023-10-29:
Hello, sorry for the late comment, but I think I have a possible solution. Basically, the bar that holds the current radar can be removed and moved down and angled into the air intake of the vehicle, and a top-mounted screw can be screwed into the bar by either making a hole in the plate from the top, or from the side (although this may be tricky due to the angle at which the face of the bar meets the plate).

[image]

[image]

[image]

[image]

[image]

[image]

SashaLuich, 2023-11-17:
https://acrobat.adobe.com/link/acrobat/combine-pdf/?x_api_client_id=adobe_com&x_api_client_location=combine_pdf&dropinId=verb-combine-pdf