Nova-UTD/hardware #7: Spec cameras

jruths, 2023-10-26:
Here is an example of an automotive/industrial grade line of cameras: [https://thinklucid.com/](https://thinklucid.com/)

meyli1, 2023-10-26:
Google doc for anybody interested in doing more research on Power Over Ethernet (PoE) cameras: https://docs.google.com/document/d/1HqdwOfh-E7vD0OC0tGWWJLxN9HN-gms8jzxGqiUcFYU/edit?usp=sharing

Camera Must:
- Be power over Ethernet
- Be able to sustain external environment (Robust hefty camera)
- Low Memory taken up: (Lower resoulution or features that allow memory compression)
- All 4 cameras should be under a budget of ~$2000

Helpful google search start links:
- Surround View System
- PoE Industrial Camera
- Memory Compression PoE Cameras

Abhi220180, 2023-11-17:
If we can get the current bandwidth that the ZED camera operates as well as the bandwidth reduction, it would smooth over the current research process.

closed 2024-02-01 by danielv012