Nova-UTD/hardware #19: Install new ZED X camera + NVIDIA AGX perception computer

jruths, 2024-06-26:


5408Andy, 2024-09-09:
### 09/09/2024 Update:

- **Status:** Doing research on the best implementation/mount for ZED cameras.
- **Next Steps:** Create rough engineering requirements, including schematics that show the exact position of the ZED cameras on the car, pitch/angle, required field of view, and the materials needed for mounting (e.g., 80/20 material).
- **Projected Completion:** TBD (subject to feedback and coordination with Gueren and Abhinav)
- **Update:** I have been exploring the optimal placement for the ZED cameras, with a focus on achieving front-facing, high positioning, wide field of view, and ensuring minimal glare or obstructions. The mounting points on the car’s front, back, and side rails. Additionally, I will need coordinate with Gueren on the ROS setup and Nvidia Orin installation, which will involve running GMSL2 cables from the cameras to the Orin. I've also reviewed the physical dimensions of the cameras and the Orin to better understand the mounting challenges..

5408Andy, 2024-09-15:
### 09/16/2024 Update:

- **Status:** Waiting on ZEDX Camera/Orin configuration.
- **Next Steps:** Take Measurements/Fabricate Orin Computer Mount
- **Projected Completion:** TBD (subject to feedback and coordination with Gueren and Abhinav and Ryan)
- **Update:** We have figured out mounting position for the Orin computer and will simply find another foot 80/20 material and build it on top of the OBC mount (specific measurements will be taken this week). We will need to determine whether or not the cameras need to at an angle to capture a good enough FOV. We will also need to add the ZEDX Cam Connector PCB Board to the Orin. We should also get an idea of how to implement the wiring.

5408Andy, 2024-09-16:
**Ideas for Mounting**

Cameras

[image]

[image]

[image]

Computer

[image]

[image]

[image]

5408Andy, 2024-09-17:
**Reminder for next update...**
- **Next Steps:** Looking into 3D Printed Mounting to more precisely hold the ZEDX Cameras. Check the printer we have in the lab and see what kind of material we have (hopefully have the kind of material that can handle the heat of the sun). Need to also create better mounting solution for the Adafruit MCU rather than keeping the plastic tub (me and Ryan also looking into 3D printed solution that hooks to the rails of the car).

5408Andy, 2024-09-23:
### 09/24/2024 Update:

- **Status:** Continued working on mount design.
- **Next Steps:** Fabricate Orin Computer Mount and Prototype 3D Print Mount
- **Projected Completion:** TBD
- **Update:** 

[image]
[image]
[image]

5408Andy, 2024-09-23:
This is a first prototype that encapsulates the cameras. Printing something like this will take a bit of material but will protect the camera. There are plenty of ways we could have optimized or gone for a completely different design entirely but..... we just wanted to go with something that will be sturdy and protect the camera.

5408Andy, 2024-10-01:
### 09/24/2024 Update:

- **Status:** Tested mount design.
- **Next Steps:** Fabricate Orin Computer Mount and continue iterating mount to create better FOV.
- **Projected Completion:** TBD
- **Update:** 

@NitroxNinja (Ryan) will continue to CAD variations of the mount but some additional 80/20 might be required to create better position for FRONT/BACK cameras.

Current prototype of mount fits camera and mounting holes need to be increased in diameter to account for shrinkage during cooling process.
[image]
[image]
[image]

Also got to see FOV for ZED cameras and will need to define what we would like to see. 360 Vision? How close to the car would you like to see? Har far should the cameras be seeing?....
[image]
[image]
[image]
[image]
[image]
[image]
[image]

5408Andy, 2024-10-01:
For some of the viewing angles, you can see the roof of the car blocks the bottom half of the view, so we will need to create higher mounting positions with 80/20 + 3D Printed material.

NitroxNinja, 2024-10-07:
### 10/6/2024 Update:

- **Status:** Tested mount design.
- **Next Steps:** Fabricate Orin Computer Mount and continue iterating mount to create better FOV.
- **Projected Completion:** TBD
- **Update:** 
We were thinking for a second, and I reduced the amount of material for the 3D print so it can print a little faster. It doesn't compromise structural integrity because it has a good enough factor or safety(assuming final product is made up of nylon), since assuming a flat surface traveling at the car's top speed(40MPH I think) it only experiences about 64.4N of force(may need to be recalculated, I am still learning how to run FEA).

I also have a design ready that is a little more angled. To do that, I have some 8020/ C channel design that could angle the camera. However we ran into a dilemma. As originally we were going to mount it higher up, but realized that blocked the LiDAR. To avoid the camera getting in the way, the decision was made to mount the camera further up. This challenges the true 360 camera system, because by 4 cameras, there may be some blind spots in between the front 2, and that could be detrimental in terms of fully accurate understanding of the environment to our developers... but we will proceed and suggest mounting the side cameras partially "more forward" to encapsulate a full 3D environment

[image]

5408Andy, 2024-10-07:
These will be later printed in black PLA and I didn't want to burn through my black PLA supply. 

I tried using the 3D printer in the lab but it was really weird for me to use. Ultimaker 3D Cura Slicer doesn't come with a preset to slice STL files for the Dremel Digilab printer(I can set up the presets on my own later, but I don't have much experience in this). I tried using the Dremel dedicated slicer but it was super slow for some reason and the program never booted up on my computer.

I will try and figure out how to use the Dremel 3D printer properly later, but for now I am just gonna use my personal printer.

5408Andy, 2024-10-08:
This is the picture of the mount using black PLA and mounted on the roof rack of the car.

[image]
[image]

NitroxNinja, 2024-10-15:
10/14/2024 Update:
- Status: New front/rear camera mounts
- Next Steps: Fabricate Orin Computer Mount and continue iterating mount to create better FOV, adjust just in case
- Projected Completion: TBD.....
- Update:
As I am still learning how to use Solidworks, me and Andy tested a variety of camera position placements and determined, that placing the camera 6 inches away from the horizontal bar is the optimal spot, so that it wouldn't capture the top of the windshield or pick up any other obstructions which will make it hard to parametrize stuff for the dev team. This also minimizes the amount of "blindspots" there are in the view as the ZedX camera captures. It is pretty unique on how much space it can acquire. 

Luckily, I was able to make a design that is pretty similar to the other one, and may require 2 prints, but I am also learning 3D printing as a whole again. This one is non-adjustable unfortunately, however it turns out we might this can be adjusted via the 8020 that will be attached to the 8020. 

[image]

5408Andy, 2024-10-19:
10/21/2024 Update:
- Status: New front/rear camera mounts printed
- Next Steps: Fabricate Orin Computer Mount, and continue iterating mount to create better FOV, adjust just in case
- Projected Completion: TBD......
- Update:
We will get these mounted on Monday before the meeting so we can get a see the FOV of the camera. The back camera still needs to be printed. The car might be too long to enable 360 degree FOV based on previous testing but, we will continue to adjust mounts as necessary.

[image]

[image]

5408Andy, 2024-10-20:
2 mounts of the one above have been printed and will be added to the car on Monday.

5408Andy, 2024-10-22:
All mounts are mounted... You can loosen the screws to adjust horizontal positioning. 

[image]
[image]
[image]
[image]
[image]

5408Andy, 2025-02-11:
2/10/2025 Update:
- Status: Successfully Mounted NVIDIA Orin AGX 
- Next Steps: Wire up camera cables and setup computer view
- Projected Completion: This week...
- Update:
Fabricated additional 8020 bar to mount on to "console frame" (metal that holds up the steering wheeling, etc.) which reinfornced OBD mount and also served as well suited mounting position for Orin.

[image]
[image]
[image]
[image]

5408Andy, 2025-02-18:
2/17/2025 Update:
- Status: Successfully Mounted NVIDIA Orin AGX 
- Next Steps: Wire up camera cables and setup computer view
- Projected Completion: This week...
- Update:
- Everything is wired, just waiting for testing/feedback on camera views before officially ziptie-ing to vehicle.