Nova-UTD/hardware #21: SLA battery protection

jruths, 2024-08-22:
Several times now, our SLA battery (the one that looks like a car battery) has been fully discharged.  I had thought we eliminated all parasitic leaks, but possibly one or more remain.  The problem with this, is that the SLA battery needs to have charge for the car to turn on and to charge the main battery with the main charger.  

It would be great to have a simple voltage monitor on the battery health of the SLA battery and either an electronic or manual switch to cut it off when it gets too low so that we never run into a fully discharged state any more.

jruths, 2024-08-26:
Maybe something like this would work:  [https://www.litime.com/products/litime-500a-battery-monitor-with-shunt](https://www.litime.com/products/litime-500a-battery-monitor-with-shunt)

5408Andy, 2025-03-03:
### **3/2/2025 Update: LVD and Battery Monitor**  

- **Status:** Researching and selecting LVD and battery monitor.  
- **Next Steps:** Measure the car’s current draw to pick the right LVD.  
- **Projected Completion:** TBD.  
- **Update:**  
  - We’re adding the **LiTime 500A Battery Monitor** to track SLA battery voltage in real-time.  
  - To prevent full discharge, we’ll install a **Low-Voltage Disconnect (LVD)** that cuts power when the battery gets too low.  
  - Since we don’t know the car’s exact current draw, we’re looking at **three LVD options**:  
    - **65A (BP-65)** 
    - **100A (BP-100)**
    - **220A (BP-220)** 
  - Next step: **Figure out how much current is drawn from the cars** to pick the right LVD.