Nova-UTD/hardware #23: 1/3 Scale Car - create equipment mount

jruths, 2024-08-26:
A tower of 80-20 needs to be fabricated to hold the computer, cameras, NVIDIA Orin, and LIDAR.

ArnavBMehta, 2024-09-10:
Status: Made a rough sketch for the mount of the 1/3 car
Next Steps: Will go to the lab tomorrow to get the exact measurements of the vehicle so I can get an exact mount.
Projected Completion: in 2-3 weeks my next work of exact measurements will be done tommorow
Update: I worked on getting sketches for the mount for the 1/3 car.

ArnavBMehta, 2024-09-10:
[image]

ArnavBMehta, 2024-09-16:
Status: Delayed. There is an issue with downloading the CAD Software
Next Steps: Either I will try downloading it on another laptop or I will start building it and see where it goes from there
Projected Completion: TBD
Update: I can come into lab tommorow and start to build the mount and if there any designs that you believe needs to be changed it can be done. The mount should be ready by this week so we can start adding the components next.

ArnavBMehta, 2024-09-24:
09/23/24 Update: 

Status: Sketch will be sent by tommorow 
Next Steps: After sketch is approved by professor I will start building the mound
Update: I can come to lab this week after the sketch has been approved so I can start cutting the 8020's

ArnavBMehta, 2024-10-01:
09/30/24 Update:

Status: This is a more fleshed out version sketch for the 1/3 mount including the different levels of the mount.
Next Steps: The next steps are to start cutting out the exact measurments of the 8020's so that I can make the mount accurately.
Update: I checked the lab and I have access to it so I was there today sketching my idea of the mount.

[IMG_4937.pdf](https://github.com/user-attachments/files/17198206/IMG_4937.pdf)