Nova-UTD/hardware #26: Eliminate the Tupperware for MCU

jruths, 2024-09-18:
Currently the MCU which commands the throttle on HailBopp sits inside a tupperware container.  I think we could try to buy/print/build a better container and mount it with more than velcro?

5408Andy, 2024-10-06:
### 10/07/2024 Update:

- **Status:** Gathered schematic drawings of MCUs and building case around it
- **Next Steps:** Continue working on the CAD drawing
- **Projected Completion:** Friday October 6th
- **Update:**  Currently working on 3D Printed mounted cases that can be screwed into the 80/20 slide through the concaves of the material. Trying to build something that can fully encapsulate the MCUs and has output/input ducts for wiring.

NitroxNinja, 2024-10-29:
10/30/2024 Update:

- Status: 1st CAD design is finished 
- Next Steps: Print the CAD design 
- Projected Completion: October 30th
- Update: By fulling encapsulating the MSU and doing some wire management, we make a safer product for our developers to work with. This can CAD can also be applied to other systems around the car as well. I do not have a picture right now as in writing this comment, but essentially it is a case that covers Raspberry Pi, and can be screwed on using nuts/standoffs. You can route wires through this box, which can be helpful.

NitroxNinja, 2024-11-04:
11/4/2024 Update:

- Status: 1st CAD design is... redoing
- Next Steps: Print the CAD design 
- Projected Completion: Tomorrow
- Update: By fulling encapsulating the MSU and doing some wire management, we make a safer product for our developers to work with. This can CAD can also be applied to other systems around the car as well. I do not have a picture right now as in writing this comment, but essentially it is a case that covers Raspberry Pi, and can be screwed on using nuts/standoffs. You can route wires through this box, which can be helpful. 
- I also realized, I needed to beef-up the design, but it will be sort of "backed-up" by the piece of 8020 so it can take some force. 
[image]
[image]

5408Andy, 2024-11-11:
11/4/2024 Update:

- Status: Miscalculated and messed up the measurements and didn't account for bonnet properly
- Next Steps: Thicken all the walls of the mounts, make it more rigid
- Projected Completion: TBD
- Update:

[image]
[image]

5408Andy, 2024-11-12:
[image]

5408Andy, 2025-03-03:
### **3/3/2025 Update: Raspberry Pi Mounting Case**

- **Status:** 1st CAD design completed, testing fit.  
- **Next Steps:** Print and test the Raspberry Pi enclosure with wiring.  
- **Projected Completion:** Tomorrow.  
- **Update:**  
  - Fully encapsulating the **MSU and Raspberry Pi** for better **protection and wire management**.  
  - This **modular case design** can be adapted for other systems in the car.  
  - The design allows **secure mounting with nuts/standoffs** and **easy wire routing**.  
  - Strengthened the structure to handle force, using **8020 as a support**.  
  - Below is a **photo of the current print progress**:  

[image]