Nova-UTD/hardware #30: 1/3 Scale Car - test motor mounts and wire rope connections

jruths, 2025-02-05:
When @stepan-antonyuk runs his arduino code to test the motors, we need to ensure the motors work.

RaviVenkat7, 2025-02-18:
### 02/17/2025 Update:

- **Status:** As of last week we found out the wires for the steering and server under the car were incorrectly connected to the arduino but the wheels turn when we send a signal to turn. 
- **Next Steps:** I think Hiba is working on a sketch for the mounts, but also we will try to find a way to generate more power from the motor.
- **Projected Completion:** Hopefully the sketches and the mount will be done by the next week (2/24/2025)
- **Update:**  N/A as of now

jruths, 2025-02-18:
https://www.docyke.com/products/servo-motor

hhllba, 2025-02-18:
Status: The current mount is comprised of several pieces, when a one-piece solution will eliminate the wiggling observed. 
Next Steps: A replacement solution that eliminates any wiggle room and utilizes a singular piece has already been designed in SolidWorks and will be machined soon. The cable adjustment is reliant on the motor mount and will be examined after installation of the new mount.
Projected Completion: Hopefully the mount will be done by the next week (2/24/2025)
Update:  New mount sketched, production soon

RaviVenkat7, 2025-02-24:
### 02/23/2025 Update:

- **Status:** Stephen wrote new Arduino Code and Debugged it, so now the throttle and steering server work with the code.
- **Next Steps:** Need to correct some of the Arduino code once its deployed, but shouldn't be a big issue. Also, we need to add a brake server, but unable to test as of now because the steering server is removed currently.
- **Projected Completion:** Hopefully in a weeks time we will be able to test the brake's server as well as put back the steering server to the car.
- **Update:**  N/A as of now

RaviVenkat7, 2025-03-03:
03/2/2025 Update:
Status: Nathan and the others machined the mount and bolted the motor to it.
Next Steps: Apparently it just needs to be bolted to the chassis. And only two more holes need to be drilled to mount it to the chassis.
Projected Completion: In 1-2 days it will be completed.
Update: N/A as of now